jbaumann/pio_and_stm32cubeide #1: TouchGFX Designer

ehtnevets, 2021-01-11:
I use TouchGFX Designer to generate UI and open with CubeMX to generate code with the setting as you specify. 

Following is the error (user name edited),
AttributeError: 'NoneType' object has no attribute 'get':
  File "~/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 168:
    env.SConscript(item, exports="env")
  File "~/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "~/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Volumes/TouchGFXProjects/stm32piotest/setup_cubemx_env_auto.py", line 198:
    ld_script_entry = tool_chain.find(

I'm almost there and thank you in advance for your help.

ehtnevets, 2021-01-12:
I changed 'MCU GCC Linker' to 'MCU G++ LInker' in the script just to get pass this. My .cproject doesn't contain 'option' subchild on 'MCU GCC Linker'.

Next error, 

FileNotFoundError: [Errno 2] No such file or directory: '/Volumes/TouchGFXProjects/stm32piodemo/lib/STLinkedResources/Buttons.cpp':. 

The directory contain softlinks to files being deleted and recreated when PlatformIO rebuilds index. Interestingly,  running setup_cubemx_env_auto.py (buy pio run) causes the index rebuild every 3s. 

If I close the workspace and reopen, the rebuild stops but keeps repeating again after a compile attempt.

I wonder what trigger the index rebuild.

ehtnevets, 2021-01-12:
I think something with MacOS symlink issue. I can't even open the file directly on MacOS terminal. ls shows the links but cat can't find the file.

jbaumann, 2021-01-12:
Are you using CubeMX or STM32CubeIDE? The script is heavily based on the information that STM32CubeIDE writes, and it might be something that CubeMX does not fully do. If you use CubeMX, then a first test would be to open the project once with STM32CubeIDE.

If this is not the case, and if you are using STM32CubeIDE, then the next step would be to send me your project so that I can have a look at the generated data, if that would be ok.

Regarding the symlinks: I don't think that the problem has to do with OSX. I'm using OSX as well. This suggests the the symlinks are not pointing to a valid file location, which again suggests that the information cannot be read from the files correctly...

ehtnevets, 2021-01-12:
Thanks for the response. I just realize TouchGFX Designer code generation is a mess. Even using cubeide, it wasn't a straight forward process. I have to add/modify paths in Eclipse project property. Additionally, the option 'Add necessary files as ref...' breaks TouchGFX Designer path organization. 

I give up. Back to the super slow bloated Eclipse with editor that sometimes takes seconds to respond to a key press. 

With the symlink, for some reason my Mac doesn't like it if the link directory is under the same directory of the file it's pointing to. I even tried with ln command on terminal.

Not OK,
./MainDir
   -->SubDir
        --> ActualFileDir
              -->thefile
        --> LinkedDir
             --> linktothefile

OK,
./MainDir
   -->SubDir1
        -->ActualFileDir
              -->thefile
  -->SubDir2
       -->LinkedDir
            -->linktothefile

jbaumann, 2021-01-13:
I just tested this. In the case you said is not ok (I used the file test.txt and linked to the same name in dir LinkedDir):

`LinkedDir % ln -s ../ActualFileDir/test.txt test.txt`

`lrwxr-xr-x  1 jbaumann  staff   25 13 Jan 17:46 test.txt@ -> ../ActualFileDir/test.txt`

Works. Editing the file works, is reflected in the linked original.

Which command did you use?

And I work with Eclipse on a Macbook from 2015 and have not yet seen these delays. This is strange.

I haven't used TouchGFX Designer yet, so I have no experience with that.

ehtnevets, 2021-01-14:
I did exactly what your script does except with ln command. My problem may be because of the samba directory. Designer only works on Windows and can't access network directory. It's a mess but it's the only GUI tool that doesn't cost $10Ks.

jbaumann, 2021-01-17:
You could run the Designer in a virtual machine that runs Windows (I personally have good experiences with Virtualbox). This way you could run the Designer on your local machine. And after running Designer you could simply copy its generated data to a directory mounted from OSX (or you could maybe export one of the directories inside the virtual machine).
But all of that is outside the scope of this project. If your STM32CubeIDE-project compiles successfully in the IDE it should be possible to transfer it to Platform.io. And if there is something I can help you with there, please open a new issue.

closed 2021-01-17 by jbaumann