jbaumann/pio_and_stm32cubeide #2: VSCode on Windows needs elevated right to be able to create `./lib/STLinkedResources` link files

olivierov, 2021-04-18:
When launched without elevated access rights, VSCode on Windows is not able to create `./lib/STLinkedResources` link files when the script is executed and generate a `Cannot create symlink in directory` error.

I don't know if the script can be executed as elevated while it's launched by a non-elevated instance of VSCode...

If it's not possible, could please mention to launch VSCode 'As Administrator' in the README.md?

Thank you

jbaumann, 2021-04-18:
Could you please check whether the problem is to create the link in the directory (e.g. by creating a link to a file in the parent directory) or whether the problem is that the target file cannot be accessed?
I think that starting the script with admin rights (by elevating it) should only be the last resort.

olivierov, 2021-04-18:
Sorry, I don't know how to do that.
But, even if the target link file is accessible, without elevating VSCode, I have this error:
`SETUP_CUBEMX Error: Cannot create symlink in directory 'lib/STLinkedResources'
While if I run VSCode with Administrator privileges, it works.

olivierov, 2021-04-18:
To avoid this pb, I've changed the property of the shortcut of VSCode

[image]

jbaumann, 2021-04-19:
Can you experiment with the access rights and the ownership of the directory? It looks as if there is some problem there that doesn't allow a normal user to create symlinks. To me it looks strange that you cannot create a symlink in a directory that you own...

olivierov, 2021-04-19:
In an administrator `cmd` console window I'm able to create a symlink using `mklink`.
In a standard user `cmd` console window I'm not allowed to create a symlink using `mklink`. It returns the following error:
`You do not have sufficient privilege to perform this operation.`

Searching more information, I found [this Microsoft page](https://docs.microsoft.com/en-us/windows/security/threat-protection/security-policy-settings/create-symbolic-links) in which we can read that for security reasons, elevating to administrator privileges is mandatory to create symbolic links.

jbaumann, 2021-05-01:
I now had the time to look at this. The explanation is very clear, Microsoft thinks that creating symbolic links is too dangerous for normal users. Thank you for the link to the Microsoft page explaining this. 
I think there are two options:
- Run VSCode with admin rights. I do not like this option because this way VSCode has by far too many rights.
- Change the user rights to allow the creation of symbolic links. This seems to be possible, but I do not know how exactly this is done.

olivierov, 2021-05-05:
>     * Change the user rights to allow the creation of symbolic links. This seems to be possible, but I do not know how exactly this is done.
My account has Administrator access right and I cannot create a symbolic link in a 'standard' user context (neither VSCode nor Powershell). I have to run the mlink command in an 'Administrator' context to be allowed to create a symbolic link.

I tried using pyuac (https://pypi.org/project/pyuac/) which sounds great... but I'm not able to setup it (I don't know Python very much).

closed 2021-10-02 by jbaumann