jbaumann/pio_and_stm32cubeide #3: Script creates incorrect link files in ./lib/STLinkedResources on Windows (VSCode)

olivierov, 2021-04-18:
When using VSCode on Windows, the <setup_cubemx_env_auto.py> script creates orphan links in `./lib/STLinkedResources` folder because STM32CubeMX add a project-based relative path for `locationURI` dependencies.

Error in VSCode:
[image]

Explorer results:
[image]

In the generated `.project` file, all `locationURI` start with the `PARENT-xx-PROJECT_LOC` string which is the project based relative path where `xx` indicates the number of levels up to the project path.

- STM32CubeMX generates `.project` file in which you can find `<locationURI>PARENT-4-PROJECT_LOC/Users/Olivier/STM32Cube/Repository/STM32Cube_FW_F1_V1.8.3/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal.c</locationURI>`
- The project is located in `D:/CloudStation/GitLab/modbus-remote_BB/modbus-remote_BB_FW`, 
- So, 4 levels of the project path have to be removed which gives `D:` as path to be used to replace the `PARENT-4-PROJECT_LOC` string (without trailing '/' and without the 'D:/Users/Olivier' the script adds).
- This gives `<locationURI>D:/Users/Olivier/STM32Cube/Repository/STM32Cube_FW_F1_V1.8.3/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal.c</locationURI>` which is correct:

[image]

[image]

If you need more information, feel free to ask.
Thank you.

Here is a .ioc and a .project files: [STLinkedResources issue.zip](https://github.com/jbaumann/pio_and_stm32cubeide/files/6331337/STLinkedResources.issue.zip)

jbaumann, 2021-04-18:
Did you set the option  „Add necessary files as reference in the toolchain project configuration file" in CubeMX? The relative paths are a problem if you do not do this.
If you did, the next question is which version of STM32CubeIDE you use? I‘m still on 1.5.1, so maybe they have introduced some changes in the newest version.

olivierov, 2021-04-18:
> Did you set the option „Add necessary files as reference in the toolchain project configuration file" in CubeMX? The relative paths are a problem if you do not do this.

It took me time to find this option, but yes, I did.

> If you did, the next question is which version of STM32CubeIDE you use? I‘m still on 1.5.1, so maybe they have introduced some changes in the newest version.

Sorry to forget to mention it. As I've just installed STM32CubeMX and STM32CubeIDE, it should be the latest version:
- STM32CubeMX: 6.2.1
- [STM32CubeIDE: 1.6.1 but I don't use it]

jbaumann, 2021-04-19:
Ok, if you set this and the problem still persists, then it might have to do with the newer version and I have to check that. First of all I have to install these versions which might need a few days...

leech001, 2021-05-02:
I have a similar problem for STM32CubeMX: 6.2.1

jbaumann, 2021-05-25:
@leech001 your approach was a good first solution, but a remaining problem is that the exact name (more precisely the digit) changes. 
I have now added a slightly different solution, added the inclusion of additional folders that are referenced inside the project (e.g. the middleware folders that you can add) and did some tests.
Please check this version and if everything works, I'll close this issue.

olivierov, 2021-05-25:
Hi @jbaumann,
Thank you for this release but, unfortunately, there is a new error line 142:

[image]

As I found a new specific case which is not taken in account (virtual folder), so I've treated it.
I've also fully treated the relative path based on project folder (PARENT-x-PROJECT_LOC).
Feel free to integrate these modifications ([setup_cubemx_env_auto.zip](https://github.com/jbaumann/pio_and_stm32cubeide/files/6542593/setup_cubemx_env_auto.zip)) in your code, but, as I don't know RegEx, I'm not able to correct your new error. Please apologize.

jbaumann, 2021-05-26:
That is an interesting error, and I hope I have fixed it. Could you test the newest version? I tried to add your virtual folder support.

olivierov, 2021-05-26:
Hi Joachim,

Good news, the modification line 146 is working well.

But there are some new errors 😉:
- line 75: do you mean `continue` instead of `next`?
- lines 76-78: removing the relative path pattern `PARENT-x-PROJECT_LOC` seems to be not enough because it leaves some old path behind, which creates an incorrect symbolic link:
  - Relative path (extract): `PARENT-4-PROJECT_LOC/Users/Olivier/STM32Cube/Repository/STM32Cube_FW_F1_V1.8.3/...`
  - `repository_location` = `D:\Users\Olivier` which is always different from `/Users/Olivier` (`startswith`)
  - Symbolic link target (extract): `D:\Users\Olivier/Users/Olivier/STM32Cube/Repository/STM32Cube_FW_F1_V1.8.3/...`
- line 190: one `tab` seems to be missing
- lines 73 to 75: it seems you are incorrectly managing the virtual folder by adding the `virtual:/virtual` string to the `virtual_dirs` list for each virtual folder encountered... which is not the virtual folder name.

So I've merged your corrections to my solution to manage the virtual folders correctly.
As in my previous version, I've also managed the relative (to project) path, because a project may reference many relative paths levels. Here are some example extracted from the project I've attached in my previous comment:
 - `PARENT-1-PROJECT_LOC/CycloneTCP_SSL_SSH_CRYPTO_Open_2_0_4/...`
 - `PARENT-4-PROJECT_LOC/Users/Olivier/STM32Cube/Repository...`

Please find the merge of our 2 solutions which works fine... (except some errors on my project side, not the script).
[setup_cubemx_env_auto.zip](https://github.com/jbaumann/pio_and_stm32cubeide/files/6549755/setup_cubemx_env_auto.zip)

PS: my name is `olivierov` not `olierov` 😁

jbaumann, 2021-05-27:
Note to self: Do not try to rush software out while focusing on your day job at the same time…

You are absolutely right, and I will never understand the way Windows, even after all this time, works with paths. I will try to find a way to unify Unix-like paths and Windows paths, but in the worst case I will simply write different cases for these two, using your code in the Windows case.

And I‘m very sorry that I wrote you name wrong, my apologies.

jbaumann, 2021-05-30:
First step: Corrected the virtual folder functionality, added a few comments. Next step is to correct the patch functionality based on your code.

jbaumann, 2021-05-31:
@olivierov Could you test the new version? It now works on my system and hopefully on yours as well.

olivierov, 2021-05-31:
Hi Joachim, virtual folders and project-based relative paths are working well, thank you.

I still have an error around line 207 of the new script. Error log extract:
```
PACKAGES: 
 - framework-stm32cubef1 1.8.3 
 - tool-ldscripts-ststm32 0.1.0 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <STLinkedResources>
Building in release mode
Compiling .pio\build\eval_f107vc\src\Core\Src\crc.o
arm-none-eabi-gcc: error: unrecognized command line option '-mruntimelibrary_c=standard_c'
Compiling .pio\build\eval_f107vc\src\Core\Src\freertos.o
Compiling .pio\build\eval_f107vc\src\Core\Src\gpio.o
Compiling .pio\build\eval_f107vc\src\Core\Src\main.o
Compiling .pio\build\eval_f107vc\src\Core\Src\stm32f1xx_hal_msp.o
arm-none-eabi-gcc: error: unrecognized command line option '-mruntimelibrary_c=standard_c'
Compiling .pio\build\eval_f107vc\src\Core\Src\stm32f1xx_hal_timebase_tim.o
arm-none-eabi-gcc: error: unrecognized command line option '-mruntimelibrary_c=standard_c'
...
```
If I change the Runtime library in STM32CubeIDE from `Standard C` to `Reduced C` (1) , the error change to:
```
Building in release mode
Compiling .pio\build\eval_f107vc\src\Core\Src\crc.o
arm-none-eabi-gcc: error: unrecognized command line option '-mruntimelibrary_c=nano_c'
...
```

Thank you again for your work, you are close to provide us a perfect cross-platform script.
Olivier.

(1)
[image]

jbaumann, 2021-06-03:
I haven't seen this problem, and I'm using "Reduced C", but I'm working with an STM32F4 instead of an STM32F1. 
PlatformIO uses a different approach to this, by setting additional values for the compiler you can choose the exact functionality you want. See this link: [https://docs.platformio.org/en/latest/platforms/ststm32.html#stm32duino-configuration-system](https://docs.platformio.org/en/latest/platforms/ststm32.html#stm32duino-configuration-system)

In the first step, I have simply removed this option and the script should print a message for every removed option. Since by default PIO selects the Nano-Spec, this should work already.

The next step would be to decide whether the selection should be explicit in the platformio.ini file or automatic. The first gives you more control, the second guarantees that both STM32CubeIDE are more in sync. The second argument is weak, since we use different compilers...

olivierov, 2021-06-07:
Hi Joachim,
Except one typo (mflag instead of m_flag), all works well. I'm now able to compile, transfer and debug on my board.
Thank you so much.
I'm also happy to have learnt a bite of Python by looking at your efficient code.

```
> Executing task: D:\Users\Olivier\.platformio\penv\Scripts\platformio.exe debug <

Processing eval_f107vc (platform: ststm32; board: eval_f107vc; framework: stm32cube)
----------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
SETUP_CUBEMX: Using the following source directories: 'Core, USB_DEVICE, Drivers, Core/Startup, Middlewares'
Removing runtimelibrary_c=standard_c
SETUP_CUBEMX: Adding the following build flags: '-mthumb, -mcpu=cortex-m3'
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/eval_f107vc.html
PLATFORM: ST STM32 (14.0.0) > STM3210C-EVAL
HARDWARE: STM32F107VCT6 72MHz, 64KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-stm32cubef1 1.8.3 
 - tool-ldscripts-ststm32 0.1.0 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <STLinkedResources>
Building in debug mode
Compiling .pio\build\eval_f107vc\src\Core\Src\crc.o
Compiling .pio\build\eval_f107vc\src\Core\Src\freertos.o
...
Compiling .pio\build\eval_f107vc\lib7de\STLinkedResources\usbd_ioreq.o
Archiving .pio\build\eval_f107vc\lib7de\libSTLinkedResources.a
Linking .pio\build\eval_f107vc\firmware.elf
Checking size .pio\build\eval_f107vc\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==========]  95.3% (used 62448 bytes from 65536 bytes)
Flash: [====      ]  40.6% (used 106520 bytes from 262144 bytes)
Building .pio\build\eval_f107vc\firmware.bin
=========================================================== [SUCCESS] Took 18.00 seconds
```

jbaumann, 2021-06-09:
Hi Olivier,

thank you very much, and thanks for all the help you provided.

And I'm looking forward to the next bug you find.

Cheers, Joachim

closed 2021-06-09 by jbaumann