jbaumann/pio_and_stm32cubeide #4: Not working with STM32CubeIDE 1.7.0 / STM32CubeMX 6.3.0

CZEMacLeod, 2021-09-21:
I don't know if it is an issue with the latest IDE, or something else, but it compiles in the CubeIDE but fails running the python script in VSCode.

`
AttributeError: 'NoneType' object has no attribute 'text':
  File "C:\Users\xxxx\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 176:
    env.SConscript(item, exports="env")
  File "C:\Users\xxxx\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\xxxx\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\xxxx\git\yyyy\zzzz\setup_cubemx_env_auto.py", line 72:
    linkedURI = linked_resource.find(".//locationURI").text
`

I had a quick look at the .project and .cproject files and don't see any elements named `locationURI`
Perhaps the node name has changed?

```xml
<linkedResources>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_cortex.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_cortex.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_dma.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_dma.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_dma_ex.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_dma_ex.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_exti.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_exti.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_flash.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_flash.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_flash_ex.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_flash_ex.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_flash_ramfunc.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_flash_ramfunc.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_gpio.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_gpio.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_i2c.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_i2c.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_i2c_ex.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_i2c_ex.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_pwr.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_pwr.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_pwr_ex.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_pwr_ex.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_rcc.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_rcc.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_rcc_ex.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_rcc_ex.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_tim.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_tim.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_tim_ex.c</name>
			<type>1</type>
			<location>D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_tim_ex.c</location>
		</link>
	</linkedResources>
```

CZEMacLeod, 2021-09-21:
Just tried converting the linked resources to relative which seems to have created locationURI nodes, but I get a different error now:
SETUP_CUBEMX Error: Unexpected relative path type 'Src/stm32g4xx_hal.c'
```xml
<linkedResources>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_cortex.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_cortex.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_dma.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_dma.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_dma_ex.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_dma_ex.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_exti.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_exti.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_flash.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_flash.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_flash_ex.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_flash_ex.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_flash_ramfunc.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_flash_ramfunc.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_gpio.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_gpio.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_i2c.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_i2c.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_i2c_ex.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_i2c_ex.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_pwr.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_pwr.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_pwr_ex.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_pwr_ex.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_rcc.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_rcc.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_rcc_ex.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_rcc_ex.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_tim.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_tim.c</locationURI>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_tim_ex.c</name>
			<type>1</type>
			<locationURI>Src/stm32g4xx_hal_tim_ex.c</locationURI>
		</link>
	</linkedResources>
	<variableList>
		<variable>
			<name>Src</name>
			<value>file:/D:/DataFiles/STM32Cube/Repository/STM32Cube_FW_G4_V1.4.0/Drivers/STM32G4xx_HAL_Driver/Src</value>
		</variable>
	</variableList>
```

jbaumann, 2021-10-02:
Sorry for the delay, I‘m currently working on another project that takes all of my spare time.
But the error message says that the type of URI is one that I haven‘t seen before. To me it looks as if the HAL files have been copied into the Workspace (which shouldn‘t happen). Is there an option in the new version of the STM32CubeIDE that does this, and if so, could you try and turn it off?
My other project will need another 2-3 weeks until I can come back and look at this…

CZEMacLeod, 2021-10-03:
The files are definitely not copied into the project folder; however, the default repository is on my D:\ drive, and the project folder is on C:\ in my example.

jbaumann, 2021-11-14:
Ok, the other project took a bit longer...

I now had the time to update the whole toolchain, both STM32CubeIDE and PlatformIO to the current versions, and I have no problems building. The entries in my .project file look like this:
```
<link>
	<name>Drivers/STM32F4xx_HAL_Driver/stm32f4xx_hal.c</name>
	<type>1</type>
	<locationURI>PARENT-6-PROJECT_LOC/Users/jbaumann/STM32Cube/Repository/STM32Cube_FW_F4_V1.26.2/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.c</locationURI>
</link>
```
Currently, I'm matching on PARENT-(\d+)-PROJECT_LOC(.*)$ to identify the prefix that I change. The changed value is the path to the STM32 repository.

In your case it looks as if the IDE has created an additional variable that is then used as a prefix. In my tests the IDE does not do this.

How did you create the project? Could you provide me with a simple step-by-step description? I would like to understand this before I try to add functionality to cope with this case.

jbaumann, 2021-12-27:
It seems the problem no longer exists. Closing the issue.

closed 2021-12-27 by jbaumann