jbaumann/pio_and_stm32cubeide #8: error of redefinition

Sheng-Xinyu, 2023-04-26:
I am a beginner with PlatformIO. Following your script, I encountered an error of redefinition during linking. Is it due to my personal reason? What are the possible solutions?

```
.pio/build/black_f407zg/src/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.o: In function `HAL_DeInit':
stm32f4xx_hal.c:(.text.HAL_DeInit+0x0): multiple definition of `HAL_DeInit'
.pio/build/black_f407zg/FrameworkHALDriver/Src/stm32f4xx_hal.o:stm32f4xx_hal.c:(.text.HAL_DeInit+0x0): first defined here
.pio/build/black_f407zg/src/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.o: In function `HAL_Init':
stm32f4xx_hal.c:(.text.HAL_Init+0x0): multiple definition of `HAL_Init'
```

jbaumann, 2023-04-26:
This looks as if somehow the framework files are used twice, which is strange. Which version of the STM32CubeIDE and PlatformIO are you using?
Does your code compile in the STM32CubeIDE? And could you maybe attach the part of your project that produces this problem (please strip it down to the absolute minimum that produces the problem)?

Sheng-Xinyu, 2023-04-26:
I generated the entire project using CubeMX, and added platformio.ini and setup_cubemx_env_auto.py in the project folder. After compilation, the project structure is as follows:
[image]
I believe that during the build process, similar target files were generated in the pio/build folder for FrameworkHALDriver and src, one from the driver generated by CubeMX and the other from PlatformIO. I am using Win11 environment with PIO version 6.1.6 and CubeMX version 6.7.0.

closed 2023-04-26 by Sheng-Xinyu

Sheng-Xinyu, 2023-04-27:
I think I found the issue. If I want to use the libraries from cubemx, I should use automatic_cubemx instead of automatic_pio!

jbaumann, 2023-04-27:
Excellent. Can I close the issue then?

Sheng-Xinyu, 2023-04-28:
Yes, and thank you for your work!