jbaumann/pio_and_stm32cubeide #10: Script fails with custom board

sidprice, 2023-11-20:
My project uses a custom board:

`{
    "build": {
        "core": "stm32",
        "cpu": "cortex-m4",
        "extra_flags": "-DSTM32G4xx -DSTM32G431xx",
        "f_cpu": "170000000L",
        "mcu": "stm32g431cbt6",
        "product_line": "STM32G431xx",
        "variant": "STM32G4xx/G431R(6-8-B)(I-T)_G441RB(I-T)"
    },
    "connectivity": [
        "can"
    ],
    "debug": {
        "default_tools": [
            "jlink"
        ],
        "jlink_device": "STM32G431CB",
        "openocd_target": "stm32g4x",
        "svd_path": "STM32G431xx.svd"
    },
    "frameworks": [
        "arduino",
        "cmsis",
        "libopencm3",
        "stm32cube",
        "zephyr"
    ],
    "name": "d_lev_io_g431cb",
    "upload": {
        "maximum_ram_size": 32768,
        "maximum_size": 131072,
        "protocol": "stlink",
        "protocols": [
            "stlink",
            "jlink",
            "cmsis-dap",
            "blackmagic",
            "mbed"
        ]
    },
    "url": "https://www.st.com/en/evaluation-tools/nucleo-g431rb.html",
    "vendor": "ST"
}`

This is the .cproject file:

`<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<?fileVersion 4.0.0?><cproject storage_type_id="org.eclipse.cdt.core.XmlProjectDescriptionStorage">
	<storageModule moduleId="org.eclipse.cdt.core.settings">
		<cconfiguration id="com.st.stm32cube.ide.mcu.gnu.managedbuild.config.exe.debug.1739418742">
			<storageModule buildSystemId="org.eclipse.cdt.managedbuilder.core.configurationDataProvider" id="com.st.stm32cube.ide.mcu.gnu.managedbuild.config.exe.debug.1739418742" moduleId="org.eclipse.cdt.core.settings" name="Debug">
				<externalSettings/>
				<extensions>
					<extension id="org.eclipse.cdt.core.ELF" point="org.eclipse.cdt.core.BinaryParser"/>
					<extension id="org.eclipse.cdt.core.GASErrorParser" point="org.eclipse.cdt.core.ErrorParser"/>
					<extension id="org.eclipse.cdt.core.GmakeErrorParser" point="org.eclipse.cdt.core.ErrorParser"/>
					<extension id="org.eclipse.cdt.core.GLDErrorParser" point="org.eclipse.cdt.core.ErrorParser"/>
					<extension id="org.eclipse.cdt.core.CWDLocator" point="org.eclipse.cdt.core.ErrorParser"/>
					<extension id="org.eclipse.cdt.core.GCCErrorParser" point="org.eclipse.cdt.core.ErrorParser"/>
				</extensions>
			</storageModule>
			<storageModule moduleId="cdtBuildSystem" version="4.0.0">
				<configuration artifactExtension="elf" artifactName="${ProjName}" buildArtefactType="org.eclipse.cdt.build.core.buildArtefactType.exe" buildProperties="org.eclipse.cdt.build.core.buildArtefactType=org.eclipse.cdt.build.core.buildArtefactType.exe,org.eclipse.cdt.build.core.buildType=org.eclipse.cdt.build.core.buildType.debug" cleanCommand="rm -rf" description="" id="com.st.stm32cube.ide.mcu.gnu.managedbuild.config.exe.debug.1739418742" name="Debug" parent="com.st.stm32cube.ide.mcu.gnu.managedbuild.config.exe.debug">
					<folderInfo id="com.st.stm32cube.ide.mcu.gnu.managedbuild.config.exe.debug.1739418742." name="/" resourcePath="">
						<toolChain id="com.st.stm32cube.ide.mcu.gnu.managedbuild.toolchain.exe.debug.1433637238" name="MCU ARM GCC" superClass="com.st.stm32cube.ide.mcu.gnu.managedbuild.toolchain.exe.debug">
							<option id="com.st.stm32cube.ide.mcu.gnu.managedbuild.option.target_mcu.1839610912" name="MCU" superClass="com.st.stm32cube.ide.mcu.gnu.managedbuild.option.target_mcu" useByScannerDiscovery="true" value="STM32G431CBTx" valueType="string"/>
							<option id="com.st.stm32cube.ide.mcu.gnu.managedbuild.option.target_cpuid.109239671" name="CPU" superClass="com.st.stm32cube.ide.mcu.gnu.managedbuild.option.target_cpuid" useByScannerDiscovery="false" value="0" valueType="string"/>
[... 190 lines cut ...]
    for config in p.get_boards().values():
  File "C:\Users\sidpr\.platformio\platforms\atmelsam\platform.py", line 105, in get_boards
    result[key] = self._add_default_debug_tools(result[key])
  File "C:\Users\sidpr\.platformio\platforms\atmelsam\platform.py", line 154, in _add_default_debug_tools
    assert openocd_chipname
AssertionError

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
`

I would appreciate help resolving this.
Thank you.

sidprice, 2023-11-20:
Hello and thank you for helping.

I have some progress but PIO does not find any source files. I selected a nucleo431G and this is the generated PlatformIO.ini file:

`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:nucleo_g431kb]
platform = ststm32
board = nucleo_g431kb
framework = stm32cube
`

I copied the script to the project root and ran:

`platformio init --board nucleo_g431kb --project-option "framework=stm32cube"
Resolving nucleo_g431kb dependencies...
Already up-to-date.
Project has been successfully updated!`

When I build I get:

`Executing task: C:\Users\sidpr\.platformio\penv\Scripts\platformio.exe run --environment nucleo_g431kb 

Processing nucleo_g431kb (platform: ststm32; board: nucleo_g431kb; framework: stm32cube)
------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_g431kb.html
PLATFORM: ST STM32 (16.1.0) > Nucleo G431KB
HARDWARE: STM32G431KBT6 170MHz, 32KB RAM, 128KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-stm32cubeg4 @ 1.4.0 
 - tool-ldscripts-ststm32 @ 0.2.0 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 23 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Linking .pio\build\nucleo_g431kb\firmware.elf
.pio\build\nucleo_g431kb\libFrameworkCMSISDevice.a(startup_stm32g431xx.o): In function `LoopFillZerobss':
(.text.Reset_Handler+0x32): undefined reference to `main'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nucleo_g431kb\firmware.elf] Error 1
================================================= [FAILED] Took 1.35 seconds =================================================
`
Sid

jbaumann, 2023-11-21:
That is understandable, because now you do not use the script at all. Please have a look at the script example in the repository:

https://github.com/jbaumann/pio_and_stm32cubeide/blob/main/automatic_cubemx/platformio.ini

and add the needed entries.

sidprice, 2023-11-21:
Sorry, I made that mistake. I copied and updated the example INI file but still I get:

`Resolving nucleo_g431kb dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing nucleo_g431kb (platform: ststm32; board: nucleo_g431kb; framework: stm32cube)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
AttributeError: 'NoneType' object has no attribute 'text':
  File "C:\Users\sidpr\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 167:
    env.SConscript(env.GetExtraScripts("pre"), exports="env")
  File "C:\Users\sidpr\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\sidpr\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\DataRoot\Projects\D-Lev\STM32Cube_Workspace\pio_import_test\setup_cubemx_env_auto.py", line 72:
    linkedURI = linked_resource.find(".//locationURI").text
========================== [FAILED] Took 0.56 seconds ==========================`

The INI file is:

`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

;[platformio]
; This is the root directory of the STM32CubeIDE project. Platformio can only
; take one source directory, thus we filter the sources using a
; source filter and additional build options to include the headers.
; This is set in the script and changing it here has no effect
;src_dir = ./
; The project headers are defined here to be available for the libraries as well
; This is no longer needed. Do not set it.
;include_dir = Core/Inc

[env:nucleo_g431kb]
; The platform name is needed to identify the right build script.
; Do not change it.
platform = ststm32
; The board must fit the uC you selected in STM32CubeIDE. Otherwise there
; will be compile errors.
board = nucleo_g431kb
; Setting the framework is not strictly necessary and might download additional
; resources that might not be needed. If you don‘t want that, change this setting
; to a comment.
; If you then have problems with compiling or linking, set it again to see whether
; it helps.
framework = stm32cube
; This is the actual script that forces platformio into compiling the
; STM32CubeIDE project.
extra_scripts = pre:setup_cubemx_env_auto.py
; The project option containing the directory in which CubeMX resides
;custom_repo_location = ~
; Set to the same toolchain as STM32CubeIDE
; For the mapping see
; https://api.registry.platformio.org/v3/packages/platformio/tool/toolchain-gccarmnoneeabi
platform_packages = toolchain-gccarmnoneeabi@1.90301.200702

; The name of the library directory in which the linked resources will
; be placed.
lib_deps = STLinkedResources
; Additional Build Flags
; Example: build_flags = -D DEBUG -D FREERTOS_TOTAL_RUNTIME_TIMER
build_flags = -D DEBUG
`

jbaumann, 2023-11-21:
Could you post your  ".project" file? There is something strange going on here. And could you verify again, that you have set "Add necessary files as reference in the toolchain project configuration file"?

sidprice, 2023-11-21:
Thank you, I can confirm the "Add necessary files as reference ..." option is selected.

.project file:
`<?xml version="1.0" encoding="UTF-8"?>
<projectDescription>
	<name>pio_import_test</name>
	<comment></comment>
	<projects>
	</projects>
	<buildSpec>
		<buildCommand>
			<name>org.eclipse.cdt.managedbuilder.core.genmakebuilder</name>
			<triggers>clean,full,incremental,</triggers>
			<arguments>
			</arguments>
		</buildCommand>
		<buildCommand>
			<name>org.eclipse.cdt.managedbuilder.core.ScannerConfigBuilder</name>
			<triggers>full,incremental,</triggers>
			<arguments>
			</arguments>
		</buildCommand>
	</buildSpec>
	<natures>
		<nature>com.st.stm32cube.ide.mcu.MCUProjectNature</nature>
		<nature>com.st.stm32cube.ide.mcu.MCUCubeProjectNature</nature>
		<nature>org.eclipse.cdt.core.cnature</nature>
		<nature>com.st.stm32cube.ide.mcu.MCUCubeIdeServicesRevAev2ProjectNature</nature>
		<nature>com.st.stm32cube.ide.mcu.MCUAdvancedStructureProjectNature</nature>
		<nature>com.st.stm32cube.ide.mcu.MCUSingleCpuProjectNature</nature>
		<nature>com.st.stm32cube.ide.mcu.MCURootProjectNature</nature>
		<nature>org.eclipse.cdt.managedbuilder.core.managedBuildNature</nature>
		<nature>org.eclipse.cdt.managedbuilder.core.ScannerConfigNature</nature>
	</natures>
	<linkedResources>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_cortex.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_cortex.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_dma.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_dma.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_dma_ex.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_dma_ex.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_exti.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_exti.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_flash.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_flash.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_flash_ex.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_flash_ex.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_flash_ramfunc.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_flash_ramfunc.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_gpio.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_gpio.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_pwr.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_pwr.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_pwr_ex.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_pwr_ex.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_rcc.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_rcc.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_rcc_ex.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_rcc_ex.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_tim.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_tim.c</location>
		</link>
		<link>
			<name>Drivers/STM32G4xx_HAL_Driver/stm32g4xx_hal_tim_ex.c</name>
			<type>1</type>
			<location>C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_tim_ex.c</location>
		</link>
	</linkedResources>
</projectDescription>
`
Sid

jbaumann, 2023-11-21:
Ok, could you please edit the script "setup_cubemx_env_auto.py" and change the line 72 from

linkedURI = linked_resource.find(".//locationURI").text

to

linkedURI = linked_resource.find(".//location").text

and report the results? 

And which version of STM32CubeIDE are you using?

sidprice, 2023-11-21:
I made your edit and built my project with the following error:

`Processing nucleo_g431kb (platform: ststm32; board: nucleo_g431kb; framework: stm32cube)
------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_g431kb.html
PLATFORM: ST STM32 (16.1.0) > Nucleo G431KB
HARDWARE: STM32G431KBT6 170MHz, 32KB RAM, 128KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-stm32cubeg4 @ 1.4.0 
 - tool-ldscripts-ststm32 @ 0.2.0 
 - toolchain-gccarmnoneeabi @ 1.90301.200702 (9.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 24 compatible libraries
Scanning dependencies...
Dependency Graph
|-- STLinkedResources
Building in release mode
Linking .pio\build\nucleo_g431kb\firmware.elf
c:/users/sidpr/.platformio/packages/toolchain-gccarmnoneeabi@1.90301.200702/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\nucleo_g431kb\libFrameworkCMSISDevice.a(startup_stm32g431xx.o): in function `LoopFillZerobss':
(.text.Reset_Handler+0x32): undefined reference to `main'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nucleo_g431kb\firmware.elf] Error 1`

I am using CubeIDE 1.13.2.

jbaumann, 2023-11-22:
Ok, this look good already. The script runs successfully, and the compiler finds all normal files. 

For the specific error see e.g., here:

https://stackoverflow.com/questions/18137814/loopfillzerobss-has-an-undefined-reference-to-main

sidprice, 2023-11-22:
No it isn't good because the project builds just fine in CubeIDE, it appears the PIO project is not correctly configured and does not find the file 'main.c'. In fact, after a 'c;lean' of the project none of the project files are compiled, only the stm library files.

jbaumann, 2023-11-22:
Just created a new project in STM32CubeIDE 1.13.2 with an STM32G431 and a custom configuration. Then I copied the python script and the ini file to that directory and changed the entries as described in the Readme.

It cleans, compiles and links without a problem:
Linking .pio/build/nucleo_g431kb/firmware.elf
Checking size .pio/build/nucleo_g431kb/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [======    ]  62.7% (used 20544 bytes from 32768 bytes)
Flash: [==        ]  15.1% (used 19824 bytes from 131072 bytes)
Building .pio/build/nucleo_g431kb/firmware.bin
================================== [SUCCESS] Took 2.79 seconds ==================================

Could you provide the exact signature of the main()-function?

And could you please try these steps with a new project?

sidprice, 2023-11-22:
As I said, this very simple Blink project builds and runs in CubeIDE.

So, I will delete and recreate the project and try from a clean starting place. I will report back later.
Thank you.

sidprice, 2023-11-22:
This is getting no better for me.

I created a new CubeMX project with minimal defintions, clock setup, SWD enable, and a single GPIO as output. I generated the project with the suggested relative library path option selected. I tested it built without error.

Next: before opening VSCode I copied the script (modified as you suggested earlier) and the PlatformIO.INI file. The latter was edited to have my board name (nucleo_g431rb) in two locations and I added "-D DEBUG" to the build flags.

When I open the folder with the new project in VSCode it went into a continuous loop, I think running the script. I had to exit VSCode to stop it.

When I opened VSCode again I got the following output:

`Resolving nucleo_g431rb dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing nucleo_g431rb (platform: ststm32; board: nucleo_g431rb; framework: stm32cube)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
SETUP_CUBEMX Error: Unexpected relative path type 'C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal.c'
========================== [FAILED] Took 0.56 seconds ==========================`

I have [zipped up this new project](https://sidprice-my.sharepoint.com/:u:/p/sid/ETJHoz-1VHhHm3cnZbKRXTUBuDeLsRyz20EfWTiDxeaQtA?e=T9jut9), hopefully this will help identify this issue. I REALLY want to be able to use PIO in VSCode, STM32CubeIDE is just not as well integrated for development as VSCode.

sidprice, 2023-11-23:
No, I am creating/building in CubeIDE :(

jbaumann, 2023-11-23:
Then I honestly have no clue, currently. I have installed a pristine version of STM32CubeIDE 1.13.2, have created a new project, added the script and the platformio.ini, and it worked. I used your project, and after opening it in my version of STM32CubeIDE, cleaning and building it, then copying the script and platformio.ini, it worked.
And I assume the library files are all installed (Help->Manage Embedded Software Packages). The version I have installed is 1.5.1.

sidprice, 2023-11-23:
I just updated CubeIDE to v1.14.0 and I will experiment further. Thank you for your efforts, maybe something will occur to you later and we can resolve this issue.

sidprice, 2023-11-25:
Hello again.

I have uninstalled the PIO VSCode extension and reinstalled it. I created a new CubeIDE project and tried to use the script to import it into VSCode. Once again it fails. It appears that the script does most of the work but fails in setting up whatever is needed for VSCode. I get this output inside VSCode:

``Resolving nucleo_g431rb dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing nucleo_g431rb (platform: ststm32; board: nucleo_g431rb; framework: stm32cube)
Verbose mode can be enabled via -v, --verbose option
SETUP_CUBEMX Error: Unexpected relative path type 'C:/Users/sidpr/STM32Cube/Repository/STM32Cube_FW_G4_V1.5.1/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal.c'
========================== [FAILED] Took 0.56 seconds ==========================``

I have no idea how to debug what is going on, since you were able to use the project I sent I have to assume there is some error in my VSCode environment. Any suggestions as to how to debug it?

Where is the "verbose" mode set, perhaps it could reveal a useful data point.

wormyrocks, 2023-12-11:
I took another look at this sample project and found the issue:
Sid, "Linked Resources" in "STM32CubeIDE -> Properties for pio_import_test -> Resource -> Linked Resources" include absolute Windows paths to the HAL C files, which are stored externally to the project directory. It looks like the PIO importer does not know how to deal with absolute paths (and it certainly failed on my machine, because I have a different home directory.)
It looks like jbaumann re-ran code generation, which switches the resource links to relative paths.
In order to resolve "undefined reference to main()" you mention [here](https://github.com/jbaumann/pio_and_stm32cubeide/issues/10#issuecomment-1819849809), it's necessary to uncomment `src_dir = ./` and `include_dir = Core/Inc` in platformio.ini.

jbaumann, 2024-01-15:
@wormyrocks thanks for debugging this (sorry for the delay, I got Covid). My problem is that I do not use Windows, but OSX, and the change to relative paths might also have to do with the underlying OS, not only with the version of STM32CubeIDE.
Did the problem go away with the new version of STM32CubeIDE or does it still exist?

closed 2024-02-18 by jbaumann