jbaumann/pio_and_stm32cubeide #11: Working with pio drivers while Cube drivers present

GregoryLakewood, 2024-01-05:
More of a question than an issue, I am not sure if I am using this right.
CubeIDE generates `Drivers/` folder, and since `.py` script includes all of the folders in `src_dir = ./` I get redefinition of symbols error.
To bypass this I needed to change `.py` setup script so it ignores CubeIDE generated folder.
Is it the intended way tho? Maybe it can be done in platform.ini for more convenience?

jbaumann, 2024-01-15:
I don't understand the situation yet. What exactly is happening? Could you describe the project layout and maybe the output of the script? And I assume we are talking about the "automatic_cubemx" script?

GregoryLakewood, 2024-01-15:
Reproduction:

1. make a project with STM32CubeIDE
2. follow instructions `pio_and_stm32cubeide/automatic_pio/Readme.MD`
3. build pio in vscode
4. get redefinion of symbols error

My first thought was to modify .py script so it ignores `Drivers/` folder.

jbaumann, 2024-01-15:
Please use the automatic_cubemx script, the others have not been changed in a long time and are mostly for documentation purposes.

GregoryLakewood, 2024-01-17:
My project requires more automatic_pio option rather than the other.
I understand that automatic_pio is not supported.
Nevertheless, simply ignoring Drivers/ folder works well in my case.