jbaumann/pio_and_stm32cubeide #14: FatFs library doesn't compile?

IndianBoy42, 2024-05-30:
```
lib/STLinkedResources/ccsbcs.c:24:10: fatal error: ../ff.h: No such file or directory
   24 | #include "../ff.h"
      |          ^~~~~~~~~
compilation terminated.
```
The include path of the fatfs library seems strange in a way that makes this not work, because the parent dir is not linked

IndianBoy42, 2024-05-30:
temporarily patched using: cp /home/amedhi/STM32Cube/Repository/STM32Cube_FW_F4_V1.28.0/Middlewares/Third_Party/FatFs/src/*.h lib/

jbaumann, 2024-08-15:
Fixing this problem took quite a while. Sorry for that.

Since the files referenced by STM32CubeIDE do not contain the .h files that might be found somewhere in the third party directories I combined the inclusion of all referenced third party source files as sources explicitly (creating the needed directories on the fly) and in addition added every directory in which one of these third party source files as another include directory.

For FatFS this works, for other third-party modules it should as well.

closed 2024-08-15 by jbaumann